spracing/betaflight #3: Will not re-arm using DSHOT600 or 1200

carloscampos775, 2019-05-16:
The quad will successfully arm once using DSHOT600 or DSHOT1200 but will not re-arm again without being power cycled. The OSD will say ARMED but the motors do not spin. Using Multishot the quad will arm and disarm without issue.

jflyper, 2019-05-21:
@carloscampos775 As always, config `diff` is required for investigation.

carloscampos775, 2019-05-21:
# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.0.0 May  7 2019 / 01:00:38 (b1d11273e) MSP API: 1.41

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 

# name
name CARLOSFPV

# resources

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP
feature -TRANSPONDER

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1825 2100 0 0
aux 1 13 1 1825 2100 0 0
aux 2 35 0 1800 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set min_throttle = 1070
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 32000
set vbat_max_cell_voltage = 435
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2432
set osd_rssi_pos = 2074
set osd_tim_2_pos = 2455
set osd_vtx_channel_pos = 2060
set osd_current_pos = 2439
set osd_mah_drawn_pos = 2447
set osd_craft_name_pos = 2048
set osd_avg_cell_voltage_pos = 2400
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = NTSC

# profile
profile 0

set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set d_pitch = 48
set f_pitch = 125
set d_roll = 45
set f_roll = 125
set f_yaw = 125
set d_min_roll = 34
set d_min_pitch = 37
set d_min_boost_gain = 30

# rateprofile
rateprofile 0

set roll_rc_rate = 77
set pitch_rc_rate = 77
set yaw_rc_rate = 137
set roll_expo = 11
set pitch_expo = 11
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 72
set tpa_rate = 30
set tpa_breakpoint = 1750

# end the command batch

carloscampos775, 2019-05-21:
@jflyper Posted Diff

hydra, 2019-05-21:
@carloscampos775 can you please try using cli `defaults` command, then just setup DSHOT and RX type and arming command and let us know if you still get the issue?

carloscampos775, 2019-05-21:
@hydra That worked. Here is the updated diff, I'll need to go one by one and enable my normal config to see what is causing the issue. 

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.0.0 May  7 2019 / 01:00:38 (b1d11273e) MSP API: 1.41

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1775 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = FIRST
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch

carloscampos775, 2019-05-21:
@hydra Looks like the issue is DSHOT Beep. Will not re-arm and the beep will not shut off

# beacon
beacon RX_LOST
beacon RX_SET

carloscampos775, 2019-05-22:
@hydra Just adding "FLIP OVER AFTER CRASH" in "MODES" to "AUX1" will cause the quad not to re-arm. This is also with Dshot Beacon disabled
[BTFL_cli_20190521_212810.txt](https://github.com/spracing/betaflight/files/3205144/BTFL_cli_20190521_212810.txt)

hydra, 2019-05-26:
@carloscampos775 great, thanks for the update, looks like there's some issue with the BF code that needs some investigration.

hydra, 2019-06-20:
Can you give us details of exactly what ESC andfirmware you're using?  As no-one else has reported this yet and we've not been able to re-produce the issue it's likely the issue may be a compatibility problem so we need more details.

carloscampos775, 2019-06-20:
They are DYS Aria BLHeli_32bit 35A ESCs and they are running 32.6. I commonly use these ESCs with other FCs and Betaflight without issue. I also use Crossfire with CRSF protocol if that is any help. This problem only seem to happen if dshop beacon or flip over after crash is enabled.

sma8t, 2019-06-20:
I also have a problem with the FC in connection with 35 A Wraith32 ESC's.
everytime i want to flash them from 32.4 to 32.6 the flashing failed. at another FC they work without any problems. At this I could flashlight synonymous easily. In addition, the FC does not recognize the ESC's the first time but only the second time. when connecting to Betaflight, however, immediately.
I also have a video on youtube. 
https://youtu.be/es5mcLKdtWk have even smoked an engine now.

orionpt, 2019-06-27:
I had the same re-arming issue and solved it by removing FLIP OVER AFTER CRASH from modes, it was configured in aux2.

Also experienced the same issues with Tekko32 35A ESCs (same firmware as the Wraith32 35A) when trying to set them up in blheli32 suite on mac and android. It would fail on first attempt and work after power cycle.

Here's my diff:

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.0.0 May  7 2019 / 01:00:38 (b1d11273e) MSP API: 1.41

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 

# name
name Orion-FPV

# resources

# mixer

# servo

# servo mix

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1050 0 0
aux 1 13 2 900 1100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_lowpass2_type = PT1
set gyro_to_use = FIRST
set dyn_lpf_gyro_max_hz = 350
set mag_hardware = NONE
set max_check = 2000
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_p_ratio = 128
set blackbox_device = SPIFLASH
set min_throttle = 1070
set dshot_idle_value = 450
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set failsafe_delay = 5
set failsafe_throttle_low_delay = 10
set vbat_scale = 109
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 3
set pid_process_denom = 1
set osd_rssi_alarm = 65
set osd_cap_alarm = 1000
set osd_vbat_pos = 2465
set osd_rssi_pos = 14394
set osd_tim_2_pos = 2488
set osd_mah_drawn_pos = 2477
set osd_craft_name_pos = 2081
set osd_display_name_pos = 293
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 225
set osd_battery_usage_pos = 2505
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set debug_mode = GYRO_SCALED
set vtx_band = 1
set vtx_power = 2
set vtx_freq = 5865
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_min_hz = 80
set dyn_lpf_dterm_max_hz = 200
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 0
set feedforward_transition = 10
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 51
set p_roll = 49
set p_yaw = 57
set d_min_boost_gain = 30

# rateprofile
rateprofile 0

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 110
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end

hydra, 2019-07-02:
@orionpt thanks for the update, hopefully I'll get some time to investigate this soon and build new binaries for you all to try based on latest code.

hydra, 2019-07-19:
new binaries are available on #1 - please test and report back.

hydra, 2019-07-27:
@orionpt @sma8t @carloscampos775 have any of you tried the updated binary yet?  would like to get some feedback.

sma8t, 2019-07-27:
@hydra.
tryed the 4.1 .bin.
the double detecting of the ESCs is still there but blheli32 firmware flashing works fine now.
I think the double detecting is caused by remaping the Motor Pins.

carloscampos775, 2019-07-31:
@hydra 
The arming issue is now gone.  DSHOT BEEP AND FLIP OVER AFTER CRASH has now working. I will continue to test the firmware

hydra, 2019-07-31:
@sma8t can you create a new issue for the double-detect esc issue and post an up-to-date diff please?

@carloscampos775 thanks for confirmation, I'll close this issue now.

closed 2019-07-31 by hydra