spracing/betaflight #5: How to flash latest Betaflight build 4.1 onto SPRacingH7EXTREME?

emski0, 2019-07-14:
The firmware loads online or offline, the board goes into DFU but nothing happens afterwards. The board doesn't flash.

Andi252, 2019-07-14:
You can copy the betaflight_4.1.0_SPRACINGH7EXTREME.bin to the sd-card and rename it to CFEXSTFW.BIN. As explained in https://github.com/spracing/betaflight/issues/1

I tried this today and the FC came up with version 4.1

I found another problem with this SD approach. It will delete the configration for the flash version. Also in the other direction from SD to flash version it will lose all the configration.

emski0, 2019-07-15:
@Andi252 , i tried flashing it using the #1 , i am getting failure patterns 3 "  = Firmware load failed. No valid firmware found".
Did you try flashing it, or just load the firmware?

Andi252, 2019-07-15:
@emski0 I just loaded the new firmware from SD, I did not try to flash it. I will try in the next few days.

hydra, 2019-07-18:
I posted some details here on facebook - https://www.facebook.com/photo.php?fbid=10158683032059829&set=a.10150089105929829&type=3

I'll repeat the post here and will update #1 ASAP.

> All SPRacingH7EXTREME and SPRacingH7NANO code now merged into betaflight/master. 🥳🎉🍾
> 
> Get coding on H7 FC's now!
> 
> `git clone git@github.com:betaflight/betaflight.git`
> `make TARGET=SPRACINGH7EXTREME`
> `make TARGET=SPRACINGH7NANO`
> 
> https://github.com/…/mast…/src/main/target/SPRACINGH7EXTREME
> https://github.com/…/…/master/src/main/target/SPRACINGH7NANO
> 
> Don't flash binaries using the configurator yet, see the flight-controller manuals and flash via dfu-util for now. Do not try flashing .hex files, you need .bin files to flash.
> 
> e.g.
> 
> `dfu-util -D obj/betaflight_4.1.0_SPRACINGH7EXTERME.bin -s 0x97CE0000:leave`
> `dfu-util -D obj/betaflight_4.1.0_SPRACINGH7NANO.bin -s 0x97CE0000:leave`
> 
> For the SPRACINGH7EXTREME you can also copy the .bin file to an SD card and upgrade the firmware using the bootloader.
> 
> Flashing sequence (read fully before starting).
> 
> * "copy betaflight_4.1.0_SPRACINGH7EXTREME.bin S:\CFEXSTFW.BIN" (where S: is your correctly formatted SD card, see BlackBox logging instructions)
> * Power off FC
> * Hold BIND button.
> * Power on FC
> * Wait for LED to flash rapidly.
> * Release BIND button to enter SP Racing Bootloader DFU mode. (Not the same as CPU ROM DFU bootloader). LED flashes slowly.
> * Press and HOLD BIND button, LED flashes ON then OFF every two second.
> * Release BIND the button after 10 flashes. Unplug FC USB cable to abort. WARNING: Releasing button after the 15th LED flash will erase the CPU requiring it to be sent back to the factory for re-programming at your expense!
> * FC will load firmware from SD card and flash external flash chip with new firmware, it happens very quickly. After flashing the led with flash quickly 10 times. Any other patterns indicates failure.
> 
> You can also use the bootloader to erase the config or downgrade if needed, see details here:
> 
> https://github.com/spracing/betaflight/issues/1
> 
> Enjoy!
> 
> - Dominic Clifton / aka Mr Cleanflight aka Mr Seriously Pro/SP Racing.

hydra, 2019-07-18:
@emski0 it is very important that you do NOT try flashing from configurator until it has been updated with H7 EXST flashing support.  It's on my todo list before BF 4.1 is released.

emski0, 2019-07-18:
@hydra tried again the sd card method : i am getting failure patterns 3 " = Firmware load failed. No valid firmware found".
The FC can load the firmware from the sd card and shows as 4.1 in cli, but it cannot flash the firmware.

hydra, 2019-07-18:
I updated #1 with DFU flashing instructions from above, please try erasing firmware before flashing, I think I forgot to add that step to the instructions, I will update them now.  Certainly if the FC boots from the file on the SD card then the firmware is valid.

hydra, 2019-07-18:
Done.  #1 updated with the ERASE step.  Please reply and let me know how you get on so I can close this issue.

emski0, 2019-07-18:
It did not work with the sd card method, however i was able to flash via dfu-util.
sometimes the board sees the sdcard, sometimes not. the same thing happens in the blackbox tab.

hydra, 2019-07-18:
what exact SD card are you using?  Perhaps it's not a good one?  I've mainly been using branded SanDisk and Samsung EVO cards.

emski0, 2019-07-18:
i did use the this formatter. All these sd card works ok with fatshark dvr recorder, frsky xlite transmitter, raspberry pi, runcam 3...
2 unbranded and 1 REMAX

hydra, 2019-07-27:
do you have links to the exact cards?  unless a developer has the exact same hardware it's very difficult to investigate?  Do you have access to other cards that you could try, if so please do and report back.

emski0, 2019-07-27:
https://www.banggood.com/Mixza-Year-of-the-Dog-Limited-Edition-C6-8GB-TF-Memory-Card-p-1294044.html?rmmds=myorder&cur_warehouse=CN
https://www.banggood.com/Remax-Original-16GB-High-Speed-TF-Card-Memory-Card-for-iPhone-Xiaomi-Mobile-Phone-p-1005879.html?rmmds=search&cur_warehouse=CN

And i have this one on its way(8GB), i will report when i receive it : https://www.banggood.com/Microdrive-8GB-16GB-32GB-64GB-128GB-C10-Class-10-High-Speed-TF-Memory-Card-With-Card-Adapter-For-Mobile-Phone-iPhone-Samsung-Huawei-Xiaomi-p-1520625.html?rmmds=myorder&ID=43653&cur_warehouse=CN

emski0, 2019-08-03:
I received the new sd card, i did few tests. So i thinks my sdcards are ok.
So if i put a firmware on the sd card as CFEXSTFW.BIN, the firmware loads on start up and i can see the Version in cli. 
However i cannot flash the firmware from the sd card, again it fails with  failure pattern 3 (3blinks) .
Also the sd card does not appear in the blackbox tab.

hydra, 2019-08-24:
I was able to flash the firmware onto your FC just fine using the instructions on #1, as repeated below.

I upgraded it from the factory firmware 20190421 to the 20190816 release.

### Firmware upgrade procedure via SD card.

* Enter SP Racing bootloader DFU mode (see below).
* Place .exst (4.0) or .bin (4.1 onwards) on SD card as CFEXSTFW.BIN

BF 4.0
`copy betaflight_4.0.0_SPRACINGH7EXTREME_EXST.exst U:\CFEXSTFW.BIN`
`copy betaflight_4.0.0_SPRACINGH7NANO_EXST.exst U:\CFEXSTFW.BIN`
`copy betaflight_4.0.0_SPRACINGH7ZERO_EXST.exst U:\CFEXSTFW.BIN`

BF 4.1:
`copy betaflight_4.1.0_SPRACINGH7EXTREME.bin U:\CFEXSTFW.BIN`
`copy betaflight_4.1.0_SPRACINGH7NANO.bin U:\CFEXSTFW.BIN`
`copy betaflight_4.1.0_SPRACINGH7ZERO.bin U:\CFEXSTFW.BIN`

where `U:` is your SD card.

* Press and HOLD BIND button.
* LED flashes ON then OFF every two second.  Release BIND the button after 7 flashes.  Remove power to FC if you need to abort.  WARNING: Releasing button after the 15th LED flash will erase the CPU requiring it to be sent back to the factory for re-programming at your expense!
* FC will erase firmware on the external flash chip, it happens very quickly.
* After erasing the led with flash quickly 10 times.  Any other patterns indicates failure.
* Wait till LED flashes slowly again.
* Press and HOLD BIND button.
* LED flashes ON then OFF every two second.  Release BIND the button after 10 flashes.  Remove power to FC if you need to abort.  WARNING: Releasing button after the 15th LED flash will erase the CPU requiring it to be sent back to the factory for re-programming at your expense!
* FC will load firmware from SD card and flash external flash chip with new firmware, it happens very quickly.
* After flashing the led with flash quickly 10 times.  Any other patterns indicates failure.

LED flashes ON then OFF every two second. Release BIND the button after 10 flashes. Remove power to FC if you need to abort. WARNING: Releasing button after the 15th LED flash will erase the CPU requiring it to be sent back to the factory for re-programming at your expense!
FC will load firmware from SD card and flash external flash chip with new firmware, it happens very quickly.
* After flashing the led with flash quickly 10 times. Any other patterns indicates failure

hydra, 2019-08-24:
[image]

Here's a photo of your board with your missing M4 pad (!) showing the SD card working.

emski0, 2019-08-24:
Ok, thanks
Closing the issue

closed 2019-08-24 by emski0