spracing/betaflight #6: Configuration lost after booting from SD

Andi252, 2019-07-14:
**Describe the bug**
The configuration of betaflight for the flash version is lost when booting the FC with a SD card where a CFEXSTFW.BIN file is located.

This bug will prevent using the firmware from SD card mode, because after any ejection of the card (e.g. during a crash) the config will be lost.

**To Reproduce**
1. change at least one parameter in betaflight configurator, save and reboot FC
2. check diff
3. shut down FC
4. insert SD card with CFEXSTFW.BIN file
5. start FC
6. check diff (no changes detected) => configuration completely lost.

**Expected behavior**
The flash version and the SD card version should have separate places to store config or use the same place, but not reset the config of the other version.

**Flight controller configuration**
no diff need

**Setup / Versions**
 - SPRACINGH7EXTREME (version 4.0.0 in flash
 -SD card with CFEXSTFW.BIN file 4.1 build from betaflight repo

hydra, 2019-07-18:
They both use the same flash sector to store config, but if there is a version mismatch then the config is reset.  This is by betaflight design.

However, it is possible to build a version of the firmware that uses the SD card to store the config.

in the target.h file look for this section:

```
#define USE_FLASH_CHIP
#define EEPROM_IN_EXTERNAL_FLASH
//#define EEPROM_IN_SDCARD
//#define EEPROM_IN_RAM
```

then change it to this:

```
#define USE_FLASH_CHIP
//#define EEPROM_IN_EXTERNAL_FLASH
#define EEPROM_IN_SDCARD
//#define EEPROM_IN_RAM
```

then build it place firmware on SD card.

Then when booting from SD card firmware the config will be stored on the SD too!

I can't remember of the top of my head if the resulting firmware will correctly ignore the config sector on the external flash or wether it will be used as an extra sector for blackbox logs on external flash.  Use cli command `flash_info` to find out.  Maybe some work there still needs to be done when building firmware.

hydra, 2019-07-27:
currently it's working as-designed.  as time progresses we can look to add more features and improve firmware/config on sdcard functionality.

closed 2019-07-27 by hydra