spracing/betaflight #8: SmartAudio V2.0 not working with LUA or OSD

carloscampos775, 2019-08-03:
SmartAudio V2.0 is not working with LUA or OSA. VTX is connected to TX4. I have confirmed that VTX is working with SmartAudio in another test quad
[H7_DIFF_SA_ISSUE.txt](https://github.com/spracing/betaflight/files/3463317/H7_DIFF_SA_ISSUE.txt)

hydra, 2019-08-16:
did you resolve this issue?  I note that your dump does not contain any `vtxtable` output do probably you just forget to configure it?

Please see https://github.com/betaflight/betaflight/wiki/VTX-tables

carloscampos775, 2019-08-24:
Hi Dominic,

That fixed the issue.

Please use this table to SmartAudio V2.0

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

hydra, 2019-08-25:
great, thanks for the update!

closed 2019-08-25 by hydra