spracing/betaflight #11: SP Racing H7 Zero current meter

Peppe911, 2020-10-05:
Hi.
I'm trying to get support for this issue, but in betaflight forum he close the issue cause i have to contact the board productor, so this is the problem:
New board , bought directly from @hydra shop, tested with only battery voltage wire from 4in1 esc, motors wire and current sensor wire.
No short circuits on the board.
ESC is HAKRC E50AX4, no issues reading current from this board with omnibus F4 V3

Voltage meter is ok.
Current meter is impossible to set.
When i measure ADC pad 2, in idle condition and when disconnected from the wire, i read about 800mv; total idle current from battery is less than 130mA.
When i connect the current sensor wire from ESC this voltage drops to about 250mv;
if i measure directly from this pad, with current sensor connected and with various loads, this voltage don't change.
If i measure directly esc current output, without connecting the wire to the FC, i read 0mv in idle, 30mv at 1.36A (4 motors at 1100) and 70mv at 2.64A (4 motors at 1150).

So i think someting is wrong in settings, or is broken in the FC or is a firmware bug.

I try also inverting resources from 
resource ADC_CURR 1 C01
resource ADC_EXT 1 C05
to
resource ADC_CURR 1 C05
resource ADC_EXT 1 C01
ADC2 PAD still in this condition.

`
# dump

# version
# Betaflight / SPRACINGH7ZERO (SP7Z) 4.2.2 Sep 14 2020 / 12:12:18 (e833ac612a) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7ZERO
manufacturer_id 

# name: x220s

# resources
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C06
resource MOTOR 8 C07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B15
resource PWM 1 B15
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
[... 1074 lines cut ...]
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
`

hydra, 2020-10-07:
As per email conversation:

The ADC inputs float when not connected, so a reading of 800mv is not unexpected, the float voltage also depends on the usage of other ADC pins.

There are NO components between the FC and the current sensor pad on the H7NANO and H7ZERO, just a PCB trace from the MCU to the ADC pads.

You can test the hardware as follows:
1. Reset the FC to defaults (via `defaults` command in CLI).
2. Connect a battery to the FC, without a battery BF will show no current usage.
3. Connect the current sensor pad to GROUND,  in the software it should read a very low current draw.
4. Disconnect the sensor pad from GROUND.
5. Use a 10k resistor and connect the current sensor pad to 3.3V pad on the FC.  In the software it should read a very high current draw.
6. Disconnect the sensor pad from 3.3V when you are finished.

If these tests perform as expected then the FC hardware is working as intended and test results and bug reports should be posted on the BF issue tracker, referencing this issue as appropriate.

Additionally, in BF there as an ADC DEBUG mode which allows you to see the raw ADC channel values for each enabled ADC pin.

hydra, 2020-10-07:
@Peppe911 reported the following results via email:

```
3.	With “1” in the current scale field, with zero offset, current is 0; as i understand, lower scale value -> higher current measured 
4.	Ok
5.	With 1000 in scale value , in idle i read 10 amps, with current pad shorted to 3.3v with the resistor i read 32.7 amps, that is a big increase with this scale. 
6.	Ok
```

Requested @Peppe911 to reset the FC to defaults, leave the current sensor configuration alone and re-test.  Don't adjust the scale and offset value when testing, the default is fine.

Peppe911, 2020-10-08:
As soon as possible i'll try, meanwhile i'm waiting for a better current meter for double check the real drain from battery.
regards

closed 2020-11-15 by stale[bot]